hkiam/HCPBridge #1: confusing explenation in README

LocalNic, 2021-06-14:
Hi, I have just a Question, your README state that this code is only working with Series4 Hoermann Motors that normaly use HCP-2 protokol.
But your README state that it simulates a UAP1 HCP which would be for older Series 3 Motors as well 

Which information is correct cause you mark the compatibility only with Series 4 in capital letters.

thomasdupas, 2021-06-16:
Hi @LocalNic , some clarification, since I did some similar reverse engineering efforts recently.
Hormann series 3 (supramatic E3 etc; but not the promatic) uses HCP1, for example with adapter board UAP1
Hormann series 4 (supramatic E4, this time also the promatic) uses HCP2, for example with adapter board UAP1-HCP
UAP1 and UAP1-HCP seem alike based on the name, but those are completely different boards.

It's also not only the protocol which is different by the way, also the pinout. 
Both are RS485 based, but 
- HCP1 has data lines on pin 5-6 and power on pin 2-3. Talks at a baudrate of 19200
- HCP2 has data lines on pin 3-4 and power on pin 1-5. Talks at a baudrate of 57600

hkiam, 2021-06-20:
It is as thomasdupas wrote, "UAP1-HCP" != "UAP1", this project simulates the UAP1-HCP adapter with the HCP2 protocol and is therefore only for Hoermann series 4.

closed 2022-02-06 by hkiam