hkiam/HCPBridge #7: Zu wenig Volt am Bus?

coreGreenberet, 2022-01-05:
Hallo,

ich habe das Ganze gemäß dem Diagram für den ESP32 verlötet. Beim Busscan vom Garagentor werden nur kurz ca 3V angelegt, was natürlich für den ESP32 zu wenig ist und da keine Antwort am Bus kommt wird er wieder abgedreht.
Ich habe leider kein Labornetzgerät umd die Volt am Wandler einzustellen, darum wollte ich das direkt am Gerät machen.

Habe ich mich hier irgendwo vertan, oder gibt es hier noch etwas zu beachten?

h07d0q, 2022-01-05:
Hab's selber noch nicht im Einsatz, aber wegen nicht die 25V auf die benötigten 3V über den Festspannungsregler auf dem Board runtertransferiert?

coreGreenberet, 2022-01-05:
naja sie sollten auf 5V, allerdings liegen die 3V bei mir am Eingang an wo eig. die 25V liegen sollten.
Vielleicht habe ich mich auch vermessen, oder dier 25V liegen nur Anfangs kurz an und dann länger die 3V falls nichts daher kommt und ich habe sie einfach nicht gesehen.

hkiam, 2022-02-06:
ich meine mich zu erinnern, dass ich zu Beginn ein ähnliches Problem hatte, da lag es am Stecker. Versuch mal mit etwas Druck bzw. klemm mal testweise was dahinter, sodass die Pins mit etwas Druck anliegen.

seppelon, 2022-02-10:
Ich habe aktuell ein ähnliches Thema.
Nach dem Einschalten des Antriebs bekommt der Bus kurz Spannung, txd auf der rs485 Platine blitzt auch einmal oder zweimal auf, dann wird die Busversorgung auch schon wieder abgeschaltet.
Ich nutze ebenfalls einen esp32, den 3,3V rs485 Wandler auf dem Foto und habe nach Schema verdrahtet. Werde das aber nochmal weiter untersuchen wo es hängt.
Am Labornetzteil versorgt kann ich mit dem Webserver kommunizieren.
Btw: der 120 Ohm Widerstand ist verzichtbar, wenn man auf der rs485 Platine bei RO eine Lötbrücke setzt, damit wird der integrierte Abschlusswiderstand aktiviert. Musste ich mangels Datenblatt auch erstmal rausfinden.

seppelon, 2022-02-13:
Konnte mein Problem lösen.
Mein Supramatic 4 hat keine Dipschalter für den Busscan. Beim Einschalten wird der Bus zwar kurz aktiv, es erfolgt aber offensichtlich kein richtiger Scan.
Über das Menü 37, Parameter 1, gefolgt von einem **langen** drücken auf Prog kann man den Scan manuell auslösen. Danach Antrieb stromlos machen und nach dem Einschalten funktioniert alles. In der Bedienungsanleitung ist leider nicht beschrieben dass man den Parameter über einen sehr langen Tastendruck aktivieren muss.

donulfo, 2022-03-03:
Hi Seppelon,
ich habe dasselbe Problem wie du. Benutze auch einen ESP32 und habe die Schaltung wie auf dem Schema verdrahtet. ESP32 blinkt kurz auf beim Busscan, und TXD auf RS485 blinkt auch kurz auf. Strom liegt an. Und auch der RS485 bekommt die korrekten 3V während des Busscans.
Hast du nach dem Busscan wie in der Anleitung beschrieben dann die 1 auf der Anzeige erhalten?

Fürs erfolgreiche kompilieren mit visual studio, musste ich nur die eine Zeile der index_html aus der main.cpp auskommentieren. Das war ja korrekt, oder ?

//#include "index_html.h"
#include "../../WebUI/index_html.h"

Für alle die auch eine SupraMatic 4 ohne die Dipschalter haben, hier die detaillierte Beschreibung wie ein Busscan durchgeführt werden kann: 

**Zubehör am SupraMatic E/P Serie 4 anmelden**
1) Ziehen Sie das Stromkabel für den Antrieb aus der Steckdose. Das anzuschließende Zubehör wird im stromlosen Zustand in den Bus-Anschluss gesteckt. Anschließend können Sie den Antrieb wieder an den Strom anschließen.

2) Rufen Sie nun über die Programmiertaste (PRG, längere Zeit gedrückt halten) das Programmiermenü (Display-Anzeige 00) auf.

3) Wählen Sie über die Pfeiltasten das Menü 37 aus. Dieses beinhaltet den Bus-Scan.

4) Anschließend tätigen Sie wieder die Programmiertaste, um den Parameter (Untermenü) 01 auszuwählen.

5) Nun drücken Sie die P-Taste erneut und halten sie so lange, bis bS im Display erscheint. Sobald sich das bS im Menü zeigt, lassen Sie die Taste los und warten Sie den Scan ab. Der wird genau jetzt durchgeführt.

6) So lange, wie der Scan läuft, wird das bS blinken. Ist der Scan beendet, ändert sich die Anzeige. Bei neu eingestecktem Zubehör zeigt sich kurzzeitig eine 1 (keine 01), bevor die Anzeige zurück auf 37 springt. Das Zubehör wurde nun erkannt und der Bus-Steckplatz eingeschaltet.

seppelon, 2022-03-05:
Hi donulfo,
zum erfolgreichen compilieren musste ich noch eine Datei mit meinen WLAN-Daten anlegen.

Ich habe die Schaltung im stromlosen Zustand angeschlossen und dann den Busscan wie von dir beschrieben ausgeführt, der endet bei mir mit 1. Danach bin ich aus dem Menü raus, musste aber nochmal den Antrieb stromlos machen, bevor dann dauerhaft Spannung am Bus war und der RS485 permanent flackert.
Warum auch immer nicht direkt nach dem Scan der Bus aktiv bleibt…

seppelon, 2022-04-04:
Findet denn überhaupt "Kommunikation" statt, also blitzt zumindest mal eine LED am RS485 Modul auf?
Anscheinend ist der Torantrieb im Rahmen seines Protokolls nicht überzeugt, dass was angeschlossen ist.
Also antwortet die Schaltung nicht (bzw. nicht wie vom Antrieb erwartet). Hast du entweder Abschlusswiderstände für den Bus verbaut oder die auf der Platine des Interaces mittels Lötbrücke aktiviert?

hesc255, 2022-04-07:
Danke für deine Antwort.
Ich habe nochmal alles geprüft:
- der von mir verbaute Widerstand zwischen A+ und B- auf dem RS485 to TTL konverter hat laut Multimeter 119,5 ohm
- Lötverbindungen  alle ok, sind noch 3 mal gemessen worden, da das Webinterface erreichbar ist, sollte es hier kein Problem geben
- Am Labornetzteil, sowie am Torantrieb (hier aber nur während des Busscans) leuchtet auf dem ESP32 und dem Stepdownwandler die entsprechenden LEDs. 
- Wenn ich am Labornetzteil resette bzw während des Busscans blitzt ganz kurz die TXD led auf dem RS485 konverter auf
-  zweites RS485 modul habe ich getestet, kein Unterschied - Verbindung zur Supramatic P4 kommt nicht zu Stande
- Versuchsaufau auf ESP8266 basis anstatt ESP32 hat ebenfalls nicht funktioniert

Bleibt wohl nur noch die Software? Ich werde nochmal von vorne beginnen und neu kompilieren

hesc255, 2022-04-07:
Ich habe jetzt nochmal weiter getestet:
- HCPBridge neu heruntergeladen, Einstellungen gemacht, Sketch kompiliert und auf den ESP32 geflasht: 
- Lötbrücke auf dem RS485 Wandler gesetzt und Widerstand nachgemessen. Den von mir gesetzten 120 ohm Widerstand habe ich entfernt
- Testweise RX und TX getauscht

Ich habe durchgängig das gleiche Problem: Busscan wird mit 0 quittiert und der Bus wird abgeschaltet. Auch der Trick von weiter oben, das der Bus erst nach einem erneuten einstecken des Torantriebs dauerhaft aktiv ist, hat bei keinem Versuch etwas geholfen.

seppelon, 2022-04-07:
Ich würde nochmal die Verbindung vom RJ12-Stecker zur RS485 Platine prüfen und da auch wirklich sicherstellen dass der richtige Pin des Steckers auf der Platine ankommt und auch Durchgang hat.
Während des Busscan muss **RXD** auf jeden Fall länger blinken/blitzen, da der Antrieb ja Signal auf den Bus gibt. Selbst wenn der ESP da nicht reagiert müsste das meiner Ansicht nach kurz zu sehen sein (man korrigiere mich, falls jemand anderer Meinung ist).

hesc255, 2022-04-08:
Die Durchgängigkeit der RJ12-Pins an die Pins  der RS485 Platine habe ich auch auf Durchgang geprüft. Ebenso die Durchgängigkeit aller Pins z.b. 

- alle Ground untereinander und gegen RJ12-Pin
- 3,3v pin des ESP32 zu plus pin RS485
- TX  pin des ESP32 zu TX pin RS485
- RX pin des ESP32 zu TX pin RS485

Spannungen nachgemessen: 
- ausgang Stepdown converter 5,01 volt
- 3,3v pin des ESP32 zu plus pin RS485 sind auch genau 3,3 volt

Eigentlich wollte ich die Platine nochmal neu erstellen, aber alle Kontakte haben durchgang und Kurzschlüsse sind auch keine vorhanden. Somit sollte das nicht nötig sein. Es könnte noch am RJ12-Stecker liegen... oder eben die Software passt nicht zum Antrieb.

egger666, 2022-04-24:
Ich hatte das gleiche Problem und bin fast wahnsinnig geworden, da der esp32 nicht mit dem supramatic reden wollte…
Meine Lösung war, dass ich den Converter in meiner Verzweiflung mal mit 5volt versorgt hatte.Danach hatte ich gleich beim 1. Versuch eine Verbindung zum Antrieb!
Laut Datenblatt kann er 5volt vertragen.

vielleicht hilft es ja auch jemandem:-)

donulfo, 2022-04-25:
Hi Egger666,
vielen Dank für diesen Tipp, das hat bei mir das Problem auch gelöst und es funktioniert nun!

hesc255, 2022-04-25:
Hi Egger666, hab tausend Dank. Den Konverter mit 5 volt zu versorgen hat mein Problem gelöst.

coreGreenberet, 2022-04-25:
Zum verrückt werden...ich hab das jetzt auch versucht und VCC vom Converter direkt an Out+ vom Spannungswandler angeschlossen.
Bei mir blinkt am Converter leider immer noch nichts =(
Mit dem Multimeter hab ich nochmal alle verbindungen geprüft. das sieht eig. in ordnung aus.
So langsam hab ich das Gefühl, dass ich beim löten den Converter geschrottet habe, oder er von Haus aus nicht ging.

Hat wer Interesse die Bridge nochmal zu basteln und bei sich zu testen und es mir dann gegen Einwurf von Münzen zukommen zu lassen?^^

hesc255, 2022-04-29:
Hi, ich hätte dir gerne geholfen, doch leider ist der esp32 bei mir nicht stabil: nach ein paar Stunden ist das webinterface nicht mehr online.

AndiP1234, 2022-05-05:
Ich habe mal angefangen eine Platine in Eagle zu entwickeln aber noch nicht bestellt.

coreGreenberet, 2022-06-04:
Es ist zwar nicht das was ich wollte, aber ich hab mir jetzt einen homee brain mit dem bisecur gateway gekauft.
Jetzt wird der status vom Tor über node-red and die loxone gepusht und loxone sendet einen webhook an den homee, welches das tor öffnet/schließt bzw. das licht ein/ausschaltet

karelkryda, 2022-06-06:
Hello,
we have a **Supramatic E4** engine and I am going to buy the necessary components for circuit no. 1 (**Arduino Pro MINI 3.3V + ESP8266 Relay**).
Will everything work or should I buy something else (see your conversations about **5V** power supply)?
(
Buy:
**Arduino Pro Mini, ATmega328P TQFP 3.3V 8MHz**
or
**Arduino Pro Mini, ATmega328P TQFP 5V 16MHz**
)

Can a standard **LAN RJ-45 cable** be used as a bus connection cable? Do you have any ideas or recommendations you came up with?

btw: 120 Ohm resistor so not necessary if I use the board (TTL to RS-485) corresponding to the picture?

Thank you very much in advance

-----------------------
I'm going to buy this:
[ESP8266 Relay](https://www.gme.cz/wifi-modul-1x-rele-5v-esp8266)
Arduino Pro MINI [3.3V](https://www.laskakit.cz/arduino-pro-mini--atmega328p-tqfp-3-3v-8mhz/) or [5V](https://www.laskakit.cz/arduino-pro-mini--atmega328-5v-16mhz--klon/)
[TTL to RS-485 converter](https://pajenicko.cz/prevodnik-max485-s-prepetovou-ochranou-ttl-na-rs-485)

AndiP1234, 2022-06-06:
> Hello, we have a **Supramatic E4** engine and I am going to buy the necessary components for circuit no. 1 (**Arduino Pro MINI 3.3V + ESP8266 Relay**). Will everything work or should I buy something [...]

R13 & R14 are 120 ohms.

I think a CAT5e should be working, it´s only 9600baud. But in my eyes using a RJ45 plug is a little dangerous, you can connect a ethernet device to your diy Hoermann bridge. Ethernet uses 3.3V and the Supramatic bus voltage is higher.

TTL to RS-485 converter
"Je kompatibilní s napájecím napětím 3.3 V a 5.0 V, a stejně tak i s úrovní signálu 3.3 V a 5.0 V"

karelkryda, 2022-06-06:
> R13 & R14 are 120 ohms.
>
> I think a CAT5e should be working, it´s only 9600baud. But in my eyes using a RJ45 plug is a little dangerous, you can connect a ethernet device to your diy Hoermann bridge. [...]

Thank you for answer,

Okay, so if I understand correctly, I don't have to buy resistors (I'll just connect 2 silver plates on R13)?

I planned to use an RJ45 cable because of the termination. On the other side of the cable, I would cut off the terminal and solder to the circuit. Which cable should I use correctly?

So if I understand the last part of the comment, does it mean that I should buy the **Arduino Pro MINI 5V**? I'm not sure if it only affects the RS-485 converter or something else that needs 3.3V.
I wonder if those 5Vs on the Arduino is the thing that was addressed above or not.

AndiP1234, 2022-06-06:
> Thank you for answer,
>
> Okay, so if I understand correctly, I don't have to buy resistors (I'll just connect 2 silver plates on R13)?
>
> I planned to use an RJ45 cable because of the termination. On [...]

R13 is not necessary. (I have adjusted my previous answer.)

I would use this, because you can use the WIFI connection of the ESP32. The voltage of the GPIOs are 3,3V, you can connect a relais board if you needs one.
ESP32 (Bisher stabil durch Nutzung beider Kerne, 25V auf 5V durch LM2596S DC-DC Step Down Module):
ESP32 (So far stable by using both cores, 25V to 5V by LM2596S DC-DC Step Down Module)
https://raw.githubusercontent.com/hkiam/HCPBridge/master/Images/esp32.png

karelkryda, 2022-06-06:
@AndiP1234
somewhere (this or another issue) I read that ESP32 stopped working after a while.

In fact, do you think circuit number 2 will be better at something?
It looks like getting an ESP32 that is in stock will be a bit of a problem :)

Btw. 1: should I use the fuse in case of circuit number 1 too?
Btw. 2: since there are 2x GND and 2x 25V on the bus cable, but only one is shown in the circuit does it mean that it is a connection of 2 cables together? So connect both GND (and 25V) cables and run them as one GND and one 25V?

AndiP1234, 2022-06-06:
wasn't the problem with an ESP12 (ESP8266)?
1. yes, use a fuse in any case. If you have a short circuit in your circuit, hopefully only the fuse will be blown. Use a <=100mA fast or super fast fuse or smaller. the information is without guarantee :)
2. yes, connect to 2x GND together, the same with the both 25V. This splits the current between the two wires. Connect the fuse after merge the two wires.

karelkryda, 2022-06-06:
it is possible, I'm not sure.

1. I'll look into it, thanks for the advice
2. great, that's what I needed to know

If anything, I'll write to you again, thank you for now 🙂

hesc255, 2022-06-14:
For me its a similar issue with the Esp32 configuration. Within the first day the webinterface is not available.

donulfo, 2022-06-14:
just checked out "resetreason" - value which is 9:

9: RTC Watch dog Reset digital core

I am now recompiling with debugging enabled to check out some more details.

karelkryda, 2022-06-14:
I hate to hear that. I just bought all the parts, put them together and in a few days I will be assembling them into the engine. 

I plan to compile the code without a web interface. It seems to me that the web interface is quite demanding and for me mainly unnecessary. 

Please get in touch if you come up with anything.

karelkryda, 2022-06-14:
@donulfo could there be a problem with the ESP32 firmware? Have you tried upgrading to the latest version?

karelkryda, 2022-06-14:
I was referring to the firmware of the ESP32 board. I think it should be possible to load new firmware. I saw a mention of the espressif/arduino-esp32 github repository.

donulfo, 2022-06-14:
[image]

i am using 3.4 -> 4.4 would be the latest one

karelkryda, 2022-06-17:
@donulfo any news?

donulfo, 2022-06-17:
Yeah , so far its still running fine with the latest libraries.

karelkryda, 2022-06-19:
@donulfo can you please help me with uploading code to ESP32? What IDE I should use? What should I install? Can you send me basic step-by-step tutorial?

Thank you in advance

donulfo, 2022-06-19:
Sure,First of all i am using visual Studio Code and platform.io as the Plugin.

There are several Tutorials out there how to Install the Tool, the libs and the IDE.
This one seems to be a good one :

https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/

Btw. After having all libs updated , my esp32 Runs fine :)

karelkryda, 2022-06-19:
@donulfo thank you. Did you also updated WebServer library? I see that the latest version is incompatible (current is 1.3.0).

donulfo, 2022-06-19:
Yes, at least i updated all in the platform.io libraries. In the platformio.ini there are still the original versions listed. But I am not the expert - so i can´t say which versions are finally used. 

But i already slightly changed the code as i wanted to have mqtt integrated... i also changed some other parts which the compiler notified me...

TUNER88, 2022-06-25:
@donulfo do you have a fork for your MQTT integration?

Have you already tested MQTT via this fork?
https://github.com/andreaswatch/HCPBridgeMqtt

I was also looking for an MQTT based solution but have not tested it yet, have to order all the parts and prototype the board.

donulfo, 2022-07-06:
> @donulfo do you have a fork for your MQTT integration?
> 
> Have you already tested MQTT via this fork? https://github.com/andreaswatch/HCPBridgeMqtt
> 
> I was also looking for an MQTT based solution but have not tested it yet, have to order all the parts and prototype the board.
> 
> Update and here is one more fork with MQTT support: https://github.com/kfroeschl/HCPBridgeHomeAssistant

Hi Tuner888,
i recently tested the 2nd MQTT implementation which seems to be much more professional than my version. The only thing i need to adjust was the following in order to get in work on my esp32:

//uncomment the following two lines (34,35)
//#define PIN_TXD 17 // UART 2 TXT - G17
//#define PIN_RXD 16 // UART 2 RXD - G16

//delete the reference to those PINs in line 235
RS485.begin(57600,SERIAL_8E1);

donulfo, 2022-07-07:
Hi Tuner888,
quick update from my side - the esp32 seems to work still fine after the first day, as i can control the door via the external switch and it´s still pingable). 
But unfortunately I am constantly loosing my mqtt connectivity , which can be easily solved with a reboot of the esp32.

snoekieboe, 2022-07-09:
I've just finished building and installing the ESP32 based solution. So far so good! I experienced the same instability with the Rs485 adapter on 3.3v. So instead i wired it for 5v and now it seems rock solid. Perhaps something to change by default in the documentation since most rs485 devices seem to be able to handle 5v just fine?

P.s. i used the fork from Andreas. Webserver, MQTT and HA discovery work just fine. I tried using the build from kfroeschl but that one seems be missing the webserver entirely. Anyone managed to get the kfroeschl fork to work with the webserver?

donulfo, 2022-07-09:
Isgreat to Heat, is Andreas fork working stable for a longer period ? Kfroeschls fork stops the mqtt Feature after a day or so

h07d0q, 2022-07-09:
Ich like to have a MQTT fork for ESP8266, but it looks like Andreas and kfroeschl only provide it for ESP32...

closed 2022-09-07 by coreGreenberet

karelkryda, 2022-10-14:
Hello,
I'm currently struggling with a problem where the door is working properly (we don't have internet in the garage yet, so the ESP32 is only connected but not used and the door is still operated by remote control). Unfortunately, after a few weeks of operation, say, the door stops responding to the remote and Bluetooth and the door has to be "rebooted" (unplugged and plugged in). The motor display shows error 07 (communication error with peripheral or add-on board).
Has anyone encountered a similar problem?

Thank you in advance

donulfo, 2022-10-18:
Hi Karel,
yes i experiences the same behaviour why i switched to the other mentioned MQTT solutions mentioned above.
With one of those the Bluetooth and remote is still working fine after a few weeks , but the MQTT service stopped working.

I don´t have a fully working solution either during these days...

Thx,
Seb

karelkryda, 2022-10-18:
Hi @donulfo,
That's strange. I wonder if the problem is with the software or the hardware. In fact, due to the lack of internet, I can't even verify the reason for the ESP32 reboot.
The biggest complication is that the gate operator motor refuses to continue working when communication with the board is interrupted and has to be turned off and on.
I have done a complete update of all libraries recently, but I have not plugged the board in at the moment and am waiting.
Technically, my problem can be solved with a remote controlled electrical outlet and simply restarting the gate when this problem occurs, but it is not an ideal solution.

Please let me know if you can figure anything out.

stomko11, 2022-12-08:
@snoekieboe trying to use fork from Andreas on my ESP32. I did modify my wifi details, host name and MQTT server details in the code, but I am struggling to understand which PINs to use for TX/RX on my ESP32. Which ones did you use?