hkiam/HCPBridge #8: Lampenstatus im WebUI bei manuellem Schalten nicht korrekt

asdt1803, 2022-05-10:
Hallo,

ich habe das Problem, dass beim manuellen Schalten der Lampe im WebUI der Status nicht korrekt angezeigt wird. Die Lampe wird geschaltet, aber der Slider springt immer wieder in die "Off"-Stellung. Auf der Statusseite steht im JSON entsprechend dauerhaft "lamp":false, egal, ob die Lampe an ist, oder nicht. Bei Torfahrt zeigen Slider  und JSON den korrekten Status.

Was kann die Ursache dafür sein?

Gifford47, 2023-02-02:
@asdt1803 hast du das problem schon gelöst? ich habe z.Z. dasselbe problem...

asdt1803, 2023-02-02:
Hallo @Gifford47 , leider nein. Mir ist nicht ganz klar, wo der Unterschied zwischen manuellem und automatischem Schalten liegt und wo der Fehler im Code liegen könnte. @hkiam scheint das Projekt leider nicht mehr aktiv zu maintainen.

Gifford47, 2023-02-02:
ich bin bei meinem [dev branch](https://github.com/Gifford47/HCPBridgeMqtt/tree/development) gerade dabei das problem zu fixen und ein paar andere dinge zu implementieren. vll findest du in naher zulunft ja die lösung

Gifford47, 2023-02-07:
@asdt1803 das problem ist nun [gefixed ](https://github.com/Gifford47/HCPBridgeMqtt/issues/5)und ist in dem o.g. dev branch zu finden

asdt1803, 2023-02-07:
Super, vielen Dank. Dann mache ich hier zu und wechsle auf deinen Branch.

asdt1803, 2023-02-07:
Fixed by @Gifford47 here: https://github.com/Gifford47/HCPBridgeMqtt/tree/development

closed 2023-02-07 by asdt1803