hkiam/HCPBridge #10: 2 Garagedoors

Handy-Harry, 2022-11-06:
First of all, thanks to everyone for this great project.
I use andreaswatch's version with the ESP32, together with Home Assistant.
Except from some wifi reception that I need to fix and for some reason I had to use UART1 because UART0 just won't work.
For the rest it works great! I even made a little PCB for it.

My problem is that I have 2 garagedoors.
My knowledge is not enough to adapt the script for simultaneous use of 2 pieces in 1 Home Assistant setup.
[image]

Can someone help me out?

MCLW, 2022-12-29:
Hi Harry, 

maybe I can help you. In my Fork I use ArduinoHa from https://github.com/dawidchyrzynski/arduino-home-assistant. It is easier to handle your devices and sensors for homeassistant. 

 Do you use a separate controller for each garage door?

Ramwells, 2023-01-12:
Hi Handy Harry,

Re controlling 2 doors, you could use Node-Red to connect to 2 controllers then "wire" the logic between them and then interface to home assistant (I have done a simple HTTP request from Node-Red to the existing code to interface with the board which returns the reqiured messages to build the logic)

I like the idea of the PCB --- couple of questions 
- what did you use to design the PCB?
- how did you get it made?
- would you share any source files please?

Ramwells, 2023-01-12:
Also - you can run Node-Red from HomeAssistant (I run HA on a Raspberry pi with Node-Red and also the Mosquitto MQTT broker.

Handy-Harry, 2023-01-12:
Yeah, I have look into it again. So many projects, so little time ;-)
The PCB is just the connections original layout I created in easyEDA and ordered by jlcpcb, but it was with the faulty TX0 and RX0. Also it was to big in the Hormann compartment.