hkiam/HCPBridge #11: Supramatic E4 Bus Scan Fails

petzi0815, 2023-01-02:
Hey. Great project! The hardware is up and running and flashed. (ESP32 Version).
When i connect to the Supramatic E4 the device is powered on for a couple of seconds, then it disconnects.

I tried to do a bus scan (Menu 37 - Long Press on 01 that starts the Bus scan).
While the bus scan is active the device is powered again and the web interface is accessible as well (display show: "waten auf verbindung".

unfortunately the bus scan seems to fail as it exits with 00 instead of 01 and the device is shut down / powered off again.

Any advisory what could be wrong in the case?

Handy-Harry, 2023-01-02:
I had the same issue. I had to use UART 1 (pin 17&16) because for some reason UART 0 didn't work. (I think something with shared usb)

petzi0815, 2023-01-02:
> I had the same issue. I had to use UART 1 (pin 17&16) because for some reason UART 0 didn't work. (I think something with shared usb)

Hey. Thanks for your quick response. Actually thats my first DIY project at all. So quite new to everything. I strictly folllowed the guidelines ;-) What do you mean with UART1 ? Is it something that needs to be soldered differently on the hardware or does the software needs to be tweaked ? Thanks in advance !

MCLW, 2023-01-07:
Did you solder the 120Ohm resistor between A and B for bus termination?

Do you see the LEDs flashing on the rs485 module while the bus scan?

Ramwells, 2023-01-11:
Hi all --- I have had the same problem as petzi0815 - has anyone come up with a solution please?

Ramwells, 2023-01-11:
Following on from my last post - in response to MCLW - I do have the 120 ohn resistor in place but there are no LEDs flashing on the RS485 module during the bus scan ---- I have triple checked the wiring and even replaced the module but no change

Ramwells, 2023-01-11:
An other follow up - I also checked the E4 bus with an oscilloscope to check a bus scan was definitely happening on Menu 37 / option 1 and signals were definitely being sent on A and B. So clearly the RS485 adapter is not responding.

Another thing I noticed is that the TX0 and RX0 pins on the ESP32 are at a constant 3.3v when the program is running .... TX2 and RX2 are at 0v ---- is it normal that TX0 and RX0 are at 3.3v??

petzi0815, 2023-01-11:
I can also confirm that the resistor was soldered. As mentioned earlier i also used the other pins as the default UART0 is depreceated - but also with no result. I would be also very happy for any further suggestions

Ramwells, 2023-01-11:
Just tried to use the SWAPUART variable (by uncommenting the //#define SWAPUART) in main - which is used to trigger a serial.swap() in the setup() routine, but it would not recompile

MCLW, 2023-01-11:
Do you connect the Rs485 Board to 3,3V or 5V? Some Boards don't work with 3,3V. Connect the board to 5V instead.

Ramwells, 2023-01-11:
Thanks Handy Harry --- Hi - I have TX->TX and RX->RX.

Thanks MCLW --- Looking at the RS485 board, the supply terminal does say VCC, not 3.3v --- so could be it needs 5V --- will double check the specs and re-try tomorrow

Ramwells, 2023-01-11:
Just checked specs - the RS485 board is apparently compatible with 3.3V and 5.0V power supply and compatible with 3.3V and 5.0V signal - however I will try with 5.0V tomorrow

Ramwells, 2023-01-11:
Handy-Harry --- what code did you use to change the pins to 17 & 16 please?

Handy-Harry, 2023-01-11:
I will look tomorrow. My project is on hold. First I have to make a better WiFi signal at my gates

Ramwells, 2023-01-11:
Thanks Handy-Harry.

I'm actually looking to possibly use an ethernet connection - so starting to play with the Olimex ESP32-PoE board - got it working over wifi, now need to add the Ethernet code..... after that I will look to add MQTT as well.... fingers crossed.

MCLW, 2023-01-12:
Check out my fork, if you look for an easy way to implement mqtt. I use ArduinoHA for full mqtt and HomeAssistant support.
[HCPBridgeHomeassitant](https://github.com/MCLW/HCPBridgeHomeassistant)

Ramwells, 2023-01-12:
Success!!! Thanks for help guys -- changing the IO to pins 16/17 worked for me:
#define RS485 Serial2  // cuse serial2 - note also RS485.begin is changed in setup
#define PIN_TXD 17 // UART 2 TXT - G17
#define PIN_RXD 16 // UART 2 RXD - G16

in setup() :
 RS485.begin(57600,SERIAL_8E1,PIN_RXD,PIN_TXD); // remap to pins 16/17

I did not need to change the supply to the RS485 adapter to 5.0v - worked fine on 3.3V

I also added code to fix the IP address

I am going to try to implement on an Olimex ESP32- gateway to give me ethernet connection instead of wifi --- will let you know if successful

Thanks MCLW for the MQTT work - will also use mqtt, although may just use the HTTP calls through node-red and convert there to MQTT first.

I'm sure all of you know about Node-Red, but if not check it out - its an incredible tool (although I am biased being an ~IBMer as I know the guys who wrote that and ~MQTT !!)

Cheers,

petzi0815, 2023-01-12:
Hey @Ramwells , congrats it's working. Could you please upload / dump your .h file with the whole source code ? Did you anything in addition to my mentioned code changes ?

So you also soldered to 16/17 instead of the tutorial version and changed the code and it's working for you? It does not for me unfortunately so you please dump your file and specify what can of HW changes you did ?

petzi0815, 2023-01-12:
... ok. i have no clue why but i was able to fix it myself. maybe i did not compile properly. i also used your bit of code. so in summary. changing the code to the following and changing the wiring to pins 16,17 instead of the tutorial pins worked for me as well. 3,3 v was okay as well.

`#define RS485 Serial2`
[..]
`#define PIN_TXD 17 // UART 2 TXT - G17`
`#define PIN_RXD 16 // UART 2 RXD - G16`
`RS485.begin(57600,SERIAL_8E1,PIN_RXD,PIN_TXD); // remap to pins 16/17`

closed 2023-01-12 by petzi0815

Ramwells, 2023-01-12:
Hi - I did move the I/O on the ESP32 to pins 16 & 17 and it worked -- as noted  by Handy Harry you need to connect TX-TX and RX-RX, so for the code below I connected TX from the RS485 board to GPIO pin 17 on the ESP32 and RX to GPIO pin 16.

Don't forget the 120 ohm resistor across the A+ & B- terminals of the RS485 board.

Once I did this and re-started the HCP bus through menu 37 option 1 the RS485 board TX/RX lights started flashing and the lifter indicated there was 1 device on the bus. 
  
The only file I changed was the main.cpp:

I also commented out the #define USERELAY, and fixed the IP address as follows:
IPAddress local_IP(192, 168, 1, 150); // Set your Static IP address
IPAddress gateway(192, 168, 1, 1); // Set your Gateway IP address
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8);   //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional
String hostname = "ESP32Jim1";

here is main.cpp in full:-
 

#include <Arduino.h>
#include <AsyncElegantOTA.h>
#include <ESPAsyncWebServer.h>
#include "AsyncJson.h"
#include "ArduinoJson.h"
#include "hciemulator.h"
//#include "index_html.h"
#include "../../WebUI/index_html.h"

/* create this file and add your wlan credentials
  const char* ssid = "MyWLANSID";
  const char* password = "MYPASSWORD";
*/
#include "../../../private/credentials.h"

// switch relay sync to the lamp
// e.g. the Wifi Relay Board U4648
//#define USERELAY

// use alternative uart pins 
//#define SWAPUART

//#define RS485 Serial //original code
#define RS485 Serial2  // changed by JE to use serial2 - note also Serial begin is changed in setup
//Added by JE to use UART2 via pins 16 and 17
#define PIN_TXD 17 // UART 2 TXT - G17
#define PIN_RXD 16 // UART 2 RXD - G16

// Relay Board parameters
#define ESP8266_GPIO2    2 // Blue LED.
#define ESP8266_GPIO4    4 // Relay control.
#define ESP8266_GPIO5    5 // Optocoupler input.
#define LED_PIN          ESP8266_GPIO2

// Hörmann HCP2 based on modbus rtu @57.6kB 8E1
HCIEmulator emulator(&RS485);

// webserver on port 80
AsyncWebServer server(80);

// called by ESPAsyncTCP-esphome:SyncClient.cpp (see patch) instead of delay to avoid connection breaks
void DelayHandler(void){
    emulator.poll();
}

// switch GPIO4 und GPIO2 sync to the lamp
void onStatusChanged(const SHCIState& state){
  //see https://ucexperiment.wordpress.com/2016/12/18/yunshan-esp8266-250v-15a-acdc-network-wifi-relay-module/
  //Setting GPIO4 high, causes the relay to close the NO contact with
  if(state.valid){    
      digitalWrite( ESP8266_GPIO4, state.lampOn ); 
[... 143 lines cut ...]
    serializeJson(root,*response);

    request->send(response);    
  });

  AsyncElegantOTA.begin(&server);
  
  server.begin();

  //setup relay board
#ifdef USERELAY
  pinMode( ESP8266_GPIO4, OUTPUT );       // Relay control pin.
  pinMode( ESP8266_GPIO5, INPUT_PULLUP ); // Input pin.
  pinMode( LED_PIN, OUTPUT );             // ESP8266 module blue L
  digitalWrite( ESP8266_GPIO4, 0 );
  digitalWrite(LED_PIN,0);
  emulator.onStatusChanged(onStatusChanged);
#endif
  
}

// mainloop
void loop(){     
  AsyncElegantOTA.loop();
}

Ramwells, 2023-01-12:
Yey ! congrats Petzi0815!!

Handy-Harry, 2023-01-12:
I'm happy you guys figured it out. For me it's some time ago since I was doing this project, so all was out of my head.
I'm very interested in the MQTT HA version. Also the wired version would be ideal for me. I have the WT32-ETH01 ESP32 in a ESPHOME setup (not for hormann) in my HA. This works perfect

Ramwells, 2023-01-14:
Hi - just FYI I have implemented a version using the Olimex ESP32 Gateway board which provides ethernet connection (instead of wifi). The serial pin assignments are slightly different as GPIO17 is used by the Ethernet function - so I re-mapped the TX of Serial2 to GPIO4 which worked. I will possibly next replace the web interface with MQTT and route through Node-Red to HA.