hkiam/HCPBridge #12: Promatic 4 flashing red

Knuppel1983, 2023-01-05:
Garage door is working, but the light on the Promatic 4 is flashing red. I also see constant communication on the RS485. Is this normal?

petzi0815, 2023-01-05:
Moin. Laut Anleitung
"Wenn die gewählte Position zu nah an der Torendlage
ZU ist, erscheint eine Fehlermeldung (LED blinkt
dauerhaft 1 × rot). Automatisch wird die Position der
Werkseinstellung eingestellt oder die zuletzt gültige
Position bleibt bestehen"

https://www.stranghoener.de/einbauanleitungen/EBA_ProMatic_4.pdf

Ggf. mal eine Lernfahrt versuchen?

... Verwendest du auch ESP32 oder einen anderen Setup? Ich versuche mich verzweifelt überhaupt daran das Teil einzurichten, aber bei mir schlägt der Bus Scan dauerthaft fehl (siehe mein Issue) - hattest du ähnliche Probleme?

pboschi, 2023-01-27:
Hi, the same happens with my Promatic 4... working but blinking red. Doesn't happen with hcpbridge removed, or using another prpject, so it should'nt be related to door calibration.

dhaude, 2023-01-30:
Hallo zusammen - ich habe mich sehr gefreut dieses Projekt auf github zu finden, da ich gerade einen neuen Hörmann ProMatic Garagentorantrieb Serie 4 BiSecur in unserer Garage verbaut habe. 
Ich konnte den ESP32 D1 ohne Fehler kompilieren und habe die doch recht einfache Schaltung aufgebaut. Wird das Board bestromt, meldet sich der ESP32 brav in meinem WLAN an - auch die OTA-Funktion kann ich aufrufen.
Wenn ich jedoch die Schaltung an den Hörmann Motor anschließe und den DIP-Switch 7 betätige wird zwar bestromt und die LED im Motor blinkt eine Weile rot/grün aber dann die Versorgungsspannung wieder weggenommen.
Das klingt ähnlich wie [petzi0815](https://github.com/petzi0815) das berichtet hat
@petzi0815 konntest du das Problem identifizieren und beheben? hast Du einen Tipp für mich?

petzi0815, 2023-01-31:
Hi @dhaude ,

2 Änderungen haben bei mir die Lösung gebracht.

a) Die PINS nicht auf die Tutorial PINs anlöten, sondern auf den PIN 16/17.
b) Den Code entsprechend an 2 Stellen anpassen:

```
#define RS485 Serial2
[..]
#define PIN_TXD 17 // UART 2 TXT - G17
#define PIN_RXD 16 // UART 2 RXD - G16
RS485.begin(57600,SERIAL_8E1,PIN_RXD,PIN_TXD); // remap to pins 16/17
```

Danach anschließen und den Bus-Scan ausführen. Menü 37 -> 01. Das Gerät bekommt dann Saft während des Scans und bleibt auch danach aktiv.

dhaude, 2023-01-31:
petzi0815 - vielen Dank für die schnelle Antwort.
Ich habe das gerade mal bei meinem Board gemacht - gebracht hat es bei mir leider nichts.
Es will einfach nicht funktionieren - dabei fiel mir auf, das während dem Bus-Scan die RxD-LED auf dem RS485 Modul dauerhaft leuchtet.
Leider ist die Garage nicht direkt am Haus, was ein Debug schwierig macht.

lugg8, 2023-06-12:
@dhaude: selbes Problem hier. Schon eine Lösung gefunden ?

hohmannc85, 2023-09-10:
Offensichtlich ist es ja ein Problem mit rx tx pin auf dem esp…. Daher würde mich interessieren welcher für das Skript genutzt wird