hkiam/HCPBridge #14: Using this project with a plain MAX485 breakoutboard

s00500, 2023-04-09:
Hi Guys,

Just got my new doors installed and made sure I could get them controlled today. I had the 6 pin connectors and some old max485 breakout boards at home. I choose to build this using only a ESP32 since I think it is powerfull enough.

When using just a max485 the direction on the bus needs to be controlled as well. The converter board shown in the readme does this automatically. This caused me some minor troubles and several hours of debugging....

Initially I could not get stuff to work. I modified the code to use Serial2 on 16/17 and added a pin that controls the max 485 direction. Several times I could get it to respond correctly.... but then the door would go into a 4 times red blink pattern till a bus scan was executed again. (I have a promatic 4)

The two things I messed up where: 
- I had added a second 120Ohm resistor, there was already one on the breakout board

- The hardware UART write operation is asyncronous. The code fills the register but that take much less time than the write. But as I controlled the Direction pin by setting it before and after the write call I screwed up the transmission. The fix here was to add a delay based on the tx buffer length before I released the direction pin again. This finally made it work perfectly fine.

I thought I post this here as an issue in case somebody runs into similar trouble. It works great for me now, and I might play with it to add it to ESPHome in some way as I prefer control via HomeAssistant (although the webui is really pretty, well done @hkiam !!!).
Hope this helps when troubleshooting

Greetings from Vienna,
Lukas

closed 2023-04-09 by s00500

MCLW, 2023-07-22:
Hi Lukas,

I have the same issue from time to time. After some hours or days, my pro Matic ran into this error. 

Can you please share your code with this fix. 

Thank you

Maik

s00500, 2023-08-01:
Hey @MCLW 
I did not work very elegant here yet haha....

I am planning to make a esphome component out of it at some point... For now if you wanna see my modified code check here:
https://git.lbsfilm.at/lbsadmin/HCPBridgeESP32