hkiam/HCPBridge #19: SoftwareSerial.cpp

hohmannc85, 2023-09-09:
ich erhalte den Fehler: SoftwareSerial.cpp:41:27: fatal error: avr/interrupt.h: No such file or directory
compilation terminated.

nachdem ich etwas gegoogled habe wird softwareSerial wohl nicht von esp32 unterstützt, dennoch ist es so in dem ordner angegeben. ich bekomm den Sketch leider nicht auf mein esp32 :(

MCLW, 2023-09-09:
Hi, 

hast du die lib plerup/EspSoftwareSerial installiert? Damit sollte es gehen.

hohmannc85, 2023-09-10:
Hey, ja hab ich! Ich kann mir vorstellen den falschen esp32 Chip. Ich Versuch es nun auch mit Visual Studio statt der arduino Software! Danke für deine Unterstützung!

hohmannc85, 2023-09-10:
> Hi,
> 
> hast du die lib plerup/EspSoftwareSerial installiert? Damit sollte es gehen.

@MCLW 
ich hab inzwischen 3 verschiedene ESP32 auf die ich den sketch mit visual studio laden konnte; das Webinterface ist da; aber leider scheint die Verbindung nicht erfolgreich, weil der bus nach einigen Sekunden den ström abstellt. Hast du noch eine Idee, was ich da machen kann?

MCLW, 2023-09-10:
Welche Pins hast du an dem Modbus Modul angeschlossen? 16/17?

hohmannc85, 2023-09-10:
@MCLW 
Ich habe jetzt einen esp bestellt der rx und tx2 auch auf 16/17 hat. Meiner hat nur ein rx tx auf 1 /3.

dh. Du machst busscan und danach war auch der Strom bei dir weg aber nach Neustart vom Motor ging es?

MCLW, 2023-09-10:
Was für ein Esp Board hast du? Ich kann mir nicht vorstellen, dass die Pins auf der Platine nicht vorhanden sind. 

Genau, nach dem Busscan musste ich meinen Motor einmal neu starten.

hohmannc85, 2023-09-10:
Ist von AZ WROOM
Naja gpoi 16/17 existieren und sind auch so beschriftet aber sonst steht da rx2 und tx2 und nach Datenblatt ist das dann 16/17. oder steht auf deinem auch 16/17 und nicht rx,tx?
Probiere ich morgen beides aus! Dank dir! Und danach probiere ich die mqtt Version;-)

MCLW, 2023-09-10:
Ich hoffe das ist das richtige Board. 

Laut PinOut existiert rx2 und tx2 an pin 16/17

[](https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Pinout_Diagram.jpg?4479111012146266271)

Ich benutze einen Wemos D1 Mini. Da ist das PinOut etwas anders.

hohmannc85, 2023-09-11:
@MCLW der wemos d1 Mini ist aber ein 8266 oder?
Hast du nur den wemos oder dann 2 Module?
Kannst du mir deine Code Anpassung mal für den wemos zeigen? Davon habe ich auch noch welche liegen. Müsste pin 2 und 16 sein oder? Das 

Heute morgen der Test an pin 1/3 also normal rx tx hat nicht funktioniert. Auch nach Motor Neustart nicht, tx hat ordentlich geleuchtet, die weboberflächle war aber auch nicht erreichbar in der Zeit. Ich muss es wohl über rx2 machen bzw 16/17.

MCLW, 2023-09-11:
Nein ich habe einen mit einem esp32 drauf. 

Hier ist mein [Code](https://github.com/MCLW/HCPBridgeHomeassistant)

Pins müssten D16 und D17 sein. Schau dir mal das PinOut vom Wemos D1 an.

hohmannc85, 2023-09-11:
@MCLW  großen Dank an dich! heute ist mein neuer ESP gekommen der RX2 und TX2 auf dem chip stehen hat. nach Anpassung des Skripts ging es sofort!

der andere ESP hat mit dem neuen code auch nicht funktioniert!

closed 2023-09-11 by hohmannc85