hkiam/HCPBridge #20: Frage: welcher ESP32?

hohmannc85, 2023-09-10:
Hallo, offensichtlich liegt es an meinem ESP 32, den ich verwende. (WROOM)
Könnt ihr mir bitte einen entsprechenden ESP 32 empfehlen, der kompatibel ist?
Danke!

werty1st, 2023-09-24:
Ich habe ein esp32 dev Kit und komme damit leider nicht weiter.
Die Pins für den Modus Adapter liegen auf rx2 und tx2. Wenn ich den Bus Scan starte blinkt die RX Led auf dem Adapter. TX leider nicht.

hohmannc85, 2023-09-24:
Ist auf dem esp rx2 und tx2 beschriftet ?
Hast du den Code entsprechend angepasst?
Ich musste den Scan 2-3 mal hintereinander machen bis er es erkannt hat

werty1st, 2023-11-07:
Der ESP32 sieht so aus:
[image]

Ich habe das rs485 Modul vorher in einem Versuchsaufbau mit einem weiteren rs485 Modul und einem Arudio getestet. Der ESP32 und der Arduino konnten sich über die rs485 Module Nachrichten schicken.
Ich bin ziemlich sicher RX2 und TX2 sind GPIO16 und GPIO17.

Auf dem Video sieht man Daten ankommen. Aber der ESP32 keine Antwort schickt.
[image]

Meine Änderungen:
[image]

[image]

[image]

[image]

Ich weiß nicht genau wie ich das Problem weiter eingrenzen kann.

werty1st, 2023-11-07:
Da ich nicht mehr wusste, warum ich die Änderungen vorgenommen habe, hab ich sie jetzt weitgehend rückgängig gemacht. Die platform.ini bleibt wie oben zu sehen. Es kann sein, dass ich wegen Fehlern beim Build unter macOS (M1) etwas rumprobiert habe. Da ich unter macOS nur Probleme hatte, bin ich dann auf eine Ubuntu ARM64 VM gewechselt. Jetzt hab ich eigentlich nur noch Serial zu Serial2 geändert und die nicht vorhanden html Datei entfernt. Das Webinterface lädt trotzdem.

Die Firmware wurde erstellt und auf den ESP32 geflasht. Am Wochenende kann ich es testen.

[image]

hohmannc85, 2023-11-08:
Besten Dank für die Info. Läuft bei mir alles inzwischen;-)