hkiam/HCPBridge #22: Error 07 Communication error

AndiP1234, 2023-12-06:
Hallo,

seit einiger Zeit kommt bei mir des Öfteren Fehler 07 (Communication error) am Antrieb, dann wird die Bewegung gestoppt oder gar nicht erst ausgeführt.
Ich benutze einen ESP32.

Kennt jemand die Problematik? Der Fehler lässt sich relativ schnell reproduzieren.
Wie kann ich am Besten debuggen?

dschliche2000, 2024-04-20:
Same here got my esp32 boards from https://shop.tynet.eu/rs485-bridge-diy-hoermann-mqtt-adapter-esp32-s3-dev-board
Today. Things work except superintend error 7 on my supramatic e4 (have tree - all the same) did you solve it?

AndiP1234, 2024-04-22:
The issue happens only with low temperatures/high humidity.

AndiP1234, 2024-11-27:
I´m using a ESP32 Uno powered by LM2596S DC-DC step down module. At the past I have set the DC voltage to around 6.2 V, yesterday I rise it to > 9 V. The power is connected to the VIN pin, so the internal voltage regulator of the ESP Uno is used. With 6.2 V the ESP has only 3.1 V at the 3V3 pins. Since then I have had no more interruptions in the history. Before every 30 - 90 minutes.
[image]
[image]